UnfinishedBusiness/XmotionFirmware #1: Schematics for THC

00alkskodi00, 2020-05-10:
hi have you a schematics for pin A0 THC...? thanks

UnfinishedBusiness, 2020-05-10:
Hey @00alkskodi00 ,

  Use two 1M ohm 1/4 watt resistors and a 0.1uF aluminum electrolytic capacitor. This assumes 50:1 scaled arc voltage from the plasma unit. Have tested only with Hypertherm and Everlast machines but others should be the same. It’s a High impedance voltage divider so it’s susceptible to ghost voltage which means it will bounce around a couple volts when the torch is off and that’s okay...

You need to be mindful that this does not isolate the the scaled voltage in anyway and that once connected to the Arduino the positive voltage from the divider inside the plasma unit is actually at the potential the work lead is at (often called the “ground” clamp), keep in mind that plasma is a DC electrode negative process. This just means that the negative voltage from the electrode is connected to the Arduino’s ground. So you have to make sure the Arduino ground net is isolated from tables chassis and everything will work fine. Also, I use 5V power from USB port of control computer to supply power to Arduino so that avoids using a power supply inside control that shares 0V and chassis ground. It’s also important to isolate the plasma machine from the mains power earth ground and supply a separate dedicated earth ground for the table which will avoid a ground loop between table and control computer that will cause unwanted instability in the measured arc voltage. If running in a shop with that’s wired well and the ground impedance is low (<15 ohms) then it’s acceptable to isolate the control computer from the shop ground and allow the machine to be connected to the shops earth ground.

Hang on, I’m trying to post a picture of the schematic from my phone...

00alkskodi00, 2020-05-10:
and wich 50:1 arc voltage use you?

UnfinishedBusiness, 2020-05-10:
It will be coming from the CPC (or CNC depend on the plasma manufacturer) port from the plasma cutter. If it’s an older cutter or a cutter not designed for CNC use then the voltage divider In the schematic I supplied won’t work. What type of cutter do you have?

00alkskodi00, 2020-05-10:
I have not yet, i ony lookng now, i want to try your firmwatre or UC100 with ltp bob

UnfinishedBusiness, 2020-05-10:
Oh okay. You should look at the Everlast machines, I’ve been really happy with them on CNC tables and they’re much less expensive than Hypertherm or any of the other big names.

UC100 and Mach3/4 would work for you but then you need a BOB and an THC unit that provides up/down signals like the price CNC or the promo units and they’re very pricey. With XmotionFirmware, everything you need software wise is free and you can use an arduino uno with a inexpensive breakout board with screw terminals

00alkskodi00, 2020-05-10:
hmmm can i see theptoho where have you that all conected ? But i dont understand what you meam: 50:1 Arc Voltage.. i mean the 50:1 Arc voltage is for example Proma THC

00alkskodi00, 2020-05-10:
i know now i am a not plasma specialist. I need the hand plasma cutter whit ARC PILOT right?

UnfinishedBusiness, 2020-05-12:
Hey @00alkskodi00,

The scaled arc voltage (50:1) will be accessed at the machines CPC/CNC interface and usually can be configured with dip switches to be 10:1, 20:1, 50:1, etc... Yes, you need to find a machine with a pilot arc starting method, High-Frequency Arc Starting is far too noisy for reliable CNC application.

00alkskodi00, 2020-05-13:
Great, thank you very much for your time

UnfinishedBusiness, 2020-05-13:
You’re welcome. Anytime time!

closed 2020-05-13 by UnfinishedBusiness