UnfinishedBusiness/XmotionFirmware #2: Instructions for compiling

aduy, 2020-05-12:
I have compiled standard grbl in arduino ide, however when I go to compile the xmotionfirmware it errors out with these. I read your instructions to compile it with platformio, but I wasnt able to figure that out, or find any info on how to do so online. I need to change some of the defaults, in particular the step enable needs to be inverted for my blackbox controller.

#error "Required HOMING_CYCLE_0 not defined."
#error "WCO refresh must be greater than one."
#error "Override refresh must be greater than zero."

UnfinishedBusiness, 2020-05-12:
Hey @aduy,

I have not tried to compile it with Arduino IDE, although I'm surprised that it throws errors...

The Platform IDE route is very easy though. Install Microsoft Visual Code Studio, it's free and lightweight. Go to plugins, install the Platform IO IDE plugin. Once that's installed you simply open the folder that contains this repository and hit the platform ide upload button which will compile then upload to Arduino.

aduy, 2020-05-12:
I was able to figure it out, had to hit the icon on the bottom left, then hit the upload button. I loaded the firmware onto the blackbox and its still not responding. I did see that it displayed the correct step/mm values though. perhaps im not using ncpilot correctly. is there a startup sequence? and how do you jog?

aduy, 2020-05-13:
Can these settings be changed through the nc pilot, the same way you would change the settings from something grbl control for grbl? with standard grbl I would send the code $4=1 to invert the enable pin.

UnfinishedBusiness, 2020-05-13:
Okay great. Use the arrow keys to jog. You can only jog one axis at a time. If the DRO or Control window has focus, the arrow keys will navigate through the buttons for keyboard operations so make sure to click one in the Gcode viewer which needs to have the “focus” to jog. 

Yes it can. Go to Edit->Machine Parameters to set everything up. Note that the invert enable parameter isn’t broke out in the machine parameters dialog because I usually don’t use the amp enable, I use Teknic clear path servos and use the 5V power from USB as the amp enable to the servos which home them selfs using their RMS torque feedback against an adjustable hard stop... But you can still use the GRBL amp enable signal and configure it through serial.

aduy, 2020-05-13:
I configured everything through serial with the same settings that work in grbl, and its still not responding.

aduy, 2020-05-13:
The one thing that is for sure different is I have not configured my machine as a dual y motor. The blackbox has jumpers that just slave the two y axis drivers together. I dont see how this would cause an issue though. I can set up the second drive to use a3,a4 for step and dir with jumpers. Ill give that a try and see if it make a difference.

UnfinishedBusiness, 2020-05-13:
Hey @aduy, when you open ncPilot, are the RX and TX led’s blinking quickly? If not, then has ncPilot has not opened the serial connection correctly. It uses port description to try to auto detect the com number (or tty device on Linux or Mac) and I suspect that the blackbox usb serial covertly probably has a different description. 

Let me if that’s the case and I’ll help you get everything talking together

aduy, 2020-05-13:
I will check again, but I believe they do blink.

aduy, 2020-05-13:
one thing I noticed is I can launch the ncpilot with the .vbs script, but not using the commands in msys2.

aduy, 2020-05-13:
Ok so I had to use a firmware that I had modified to disable the clear alarm when started. the rx and tx lights are blinking.

UnfinishedBusiness, 2020-05-13:
>one thing I noticed is I can launch the ncpilot with the .vbs script, but not using the commands in msys2. it errors out saying: C:/Xmotion/Xkernel/Xkernel: error while loading shared libraries: glfw3.dll: cannot open shared object file: No such file or directory ...

@aduy - PATH=$PATH:/mingw64/bin will fix this if you launch it via the msys2 terminal. 

>Ok so I had to use a firmware that I had modified to disable the clear alarm when started. the rx and tx lights are blinking.

@aduy - If they are blinking then ncPilot is communicating with it. You should see numbers on the DRO and should be able to jog it. What does the Status say?

aduy, 2020-05-13:
The numbers on the dro change, but the motors are not even holding or moving.

aduy, 2020-05-13:
aha! so i checked with a multimeter, and with xmotionfirmware the enable pin was reading 0v relative to ground. I flashed the stock blackbox firmware and it reads 5v. no matter what settings i change in the xmotionfirmware src code it wont enable the steppers. 

Is it possible that when you simplified grbl you removed the part of the code that sends the enable signal? I may try jumpering it to 5v and see if it moves.

UnfinishedBusiness, 2020-05-13:
Oh yeah, I remember that I ended up stripping the amp enable code from the firmware to make room for the THC code, unfortunately program flash memory is practically all used up... A pull up resistor to pull that pin high would do the trick or set the pin high in code at the beginning of main function if you’re privy to direct port manipulation on the ATmega328p

aduy, 2020-05-13:
I wouldnt know what to change.

UnfinishedBusiness, 2020-05-13:
What pin is the enable on the black box?

aduy, 2020-05-13:
its pin 12 on the chip itself, also says PB0(ICP)

UnfinishedBusiness, 2020-05-13:
Okay 12 is PB4 so `PORTB |= (1 << PB4)` to set that bit

aduy, 2020-05-13:
// Start Grbl main loop. Processes program inputs and executes them.
    
    PORTB |= (1 << PB4)
    PORTB &= ~(1 << PB0); //Set torch pin off
    z_step_delay = cycle_frequency_from_feedrate((12.0f / 60.0f));
    protocol_main_loop();

  }
  return 0;   /* Never reached */
}

Like this?

aduy, 2020-05-14:
Also how do you get PB4 from that? I would think its PB0, based on the data sheet diagram of the pinout.

UnfinishedBusiness, 2020-05-14:
Can you attach a picture of the black box schematic here so I can see for sure what pin? Where you put it is correct, you just need a semicolon at the end. But i see that PB0 is the pin I’m using for the torch contactor so PB0 isn’t the amp enable pin... I must just looked online for a GRBL pin out but it’s probably dual drive pin out that’s configured here

aduy, 2020-05-14:
Ah, I see, yeah its PB0 for the enable. It looks like you are using pb0 for the torch on/off.
[image]

aduy, 2020-05-14:
heres the other board, they are connected by the large connector 
[image]

aduy, 2020-05-14:
I think the torch on off needs to be pb3 not pb0 for the blackbox

aduy, 2020-05-14:
Ok I changed the code in the cpu_map.h here

  // NOTE: USE_SPINDLE_DIR_AS_ENABLE_PIN not supported with dual axis feature.
        #define SPINDLE_ENABLE_BIT    3  // Uno Digital Pin 11
      #else
        #define SPINDLE_ENABLE_BIT    3  // Uno Digital Pin 12 $$$changed from 4 to 4
      #endif

and then changed the code in the main.c

 PORTB |= (1 << PB0); //set enable pin on
    PORTB &= ~(1 << PB3); //Set torch pin off
    z_step_delay = cycle_frequency_from_feedrate((12.0f / 60.0f));
    protocol_main_loop();

now it moves around!

do you have the pinout that you are using for your setup, so i can compare them and see what other changes need to be made? i havent tested the torch relay, but i think what i did should make that work as well.

UnfinishedBusiness, 2020-05-14:
Yeah you’re right. PB0 is the enable. Add the PB0 bit set code to main and you’re gonna be able to jog. I looked at the cpu_map.c and it looks like I made a mistake but had obviously has no consequence to the operation of the system. The bit clear of PB0 isn’t turning the torch off (as the comment states) is actually just setting the enable pin low. Change that bit clear to a bit set and you should be up and running

aduy, 2020-05-14:
You are using pb3 to fire the torch correct?

aduy, 2020-05-14:
Ok so everything is working for the most part. I figured out ncpilot only accepts g0 and g1 gcodes, so no circular moves, which isnt a huge deal, but would be nice to have. the part thats still not working is the a0 pin is reading some sort of fluctuation, but not actually reading the voltage. I dont have a variable power supply, so maybe I can rig up a AA battery to see if its reading correctly. Hopefully i soldered onto the right pin lol.

Oh and one other thing, when I hit stop in the control nothing happens, it just keeps going, and i have to turn the power off and restart ncpilot.

UnfinishedBusiness, 2020-05-14:
Okay great. So ncPilot does only render rapids and line segments in the viewer. The controller does support clockwise and clockwise arcs but I found instability when sheetcam posted arc and GRBL would whip a whole circle someplace it’s not supposed to. You can still cut arcs and circles, they just broke into line segment the same way GRBL does it internally is just done by the cam software instead. If you use the post processor included for sheetcam, it’s already setup to post line segments and handle the touch off routine

Regarding the stop button - I never use it and likely forgot to attach the abort command to it. Hit escape on the keyboard to abort anytime, think of that as your global stop button

closed 2020-05-15 by UnfinishedBusiness