UnfinishedBusiness/XmotionFirmware #4: Stepper not working

statiealexandru, 2020-10-23:
Hello!

I am running the software on an Arduino Uno board with CNC shield V3 (clone). The firmware has been flashed using VS Code.

The boards, driver and stepper run good with GRBL 1.1h, i've done the test before raising the issue.

I can't make the motor move with ncPilot nor with UniversalGcodeSender.

No error appears in ncPilot and the coordinates seem to increment. 

No error appears in UniversalGcodeSender, the parameters appear correct when exeecuting the gcode: Status RUN , IN_MOTION true and the axis coordinates incrementing.

I have compared the config file with the default grbl config and there seems to be no change.

How can i troubleshoot this further?

Thanks!
Alex

UnfinishedBusiness, 2020-10-24:
Hello @statiealexandru,

That's interesting. The pinouts seem to be the same for the step and direction pins on the CNC shield as they are on my proprietary board. Universal Gcode sender won't work with XmotionFirmware because it doesn't understand the CRC protocol that I put in place to check for serial miscommunication...

If ncPilot is incrementing the DRO then it's generating step pulses, have you modified the config.h at all from what sit's in the repository as the default config? We just need to figure out why the step pulses are not getting to the stepper drivers. If the config is all correct and you can measure step pulses at the correctly corresponding pin on your stepper drivers, then the problem is likely that they require an inverted step pulse or that they don't support such a long pulse. ncPilot by default ups step pulse time to 50 microseconds because big industrial drivers that are opto-coupled tend to like a longer pulse and it makes the system less susceptible to noise. Small digital drivers that are not opt-coupled may reject step pulses that long. If that's the case with your setup, I can walk you through how to change it because it's not just a case of modifying the $2 parameter in the case of XmotionFirmware...

Thanks,
Travis

statiealexandru, 2020-10-25:
Hey Travis!

First of all, thank you for your help! My issue is solved.

For this test I was using the cheapest driver, the A4988. I switched to another driver and everything is working well.

On the whole framework: i've discovered the CRC check within the Xkernel code. I was wondering why that is needed, now it makes sense. 

I have "adopted" your framework because i wanted to build something similar and it does not really makes sense to recreate everything if it's already built. Indeed there is no similar "lightweight" software for plasma cutters out there and retrofitting with LinuxCNC involves the MESA costs and the dedicated linux PC. 

The ncPilot code is very straightforward and the Xkernel code seems to be the same; i have only studied the motion-control file, however i feel that's where i'm going to make most of the changes if needed.

One thing that i can't figure out, however, is the firmware code. How did you make it compile in VScode (GRBL is typically flashed with Arduino Studio) and where did you add your code? I only found the xmotion.scpost file that seems to have some custom code and i can't really figure out how it works. 

If you could please give me a bit of guidance on your project structure, it would be much easier for me to adopt this project and contribute to it. 

Thanks you and great code!
Alex

UnfinishedBusiness, 2020-10-27:
Hey @statiealexandru,

Sorry for getting back a few days later... Glad you got it sorted out!

I've tried to make things as simple and robust as possible but the documentation side of things it very lacking...

So i'm using platform.io to compile XmotionFirmware (Forked from GRBL and modified for Plasma). If you install the platform io addon into VS Code, you just hit compile and upload and it takes care of the rest. I added a build of the code into main.c but there is also changes throughout in the protocal.c, system.c, etc. You can look at the git log to see what's been changed and how. The Arc voltage measurement and ATHC is all implemented in main.c though.

The Xmotion.scpost is just a post processor for SheetCAM, a popular CAM software used for Jet cutting and routing. If you use that post processor the system is pretty much turn key once the machine parameters are set up.

Let me know if you have any other question, happy to help

Thanks,
Travis Gillin

closed 2020-10-27 by UnfinishedBusiness