UnfinishedBusiness/XmotionFirmware #6: Does this work on raspberry pi?

cadrion77, 2021-05-17:
Hi there... I've been working on a plasma table for a bit and am inching closer to completion.  I see that emotion runs on linux- can it be run on a raspberry pi?  I have gotten it all to work on windows, but am planning for my final solution. Thanks for the help and your project.

UnfinishedBusiness, 2021-05-24:
Hello, Sorry for the slow reply. I'm glad you got it all up and running. The XmotionFirmware itself does not run on Raspberry Pi but I do have a new version of ncPilot that does run on the RPi 4 which then can connect via USB to Xmotion Firmware running on an Arduino Uno. Let me know if that's what you meant by it running on a RPi.

cadrion77, 2021-05-25:
Not a problem, we are all busy :)  Yes, running ncPilot was really what I was after.  I have been building a small plasma table and found your project.  I had a plan to run GRBL, but had to make a small board with relays, but this seemed like an easier way to go.  I'd be interested in anything you could let me know about ncPilot on RPi- while I have a pc that I could use, RPi would be much more portable.  I have a RPi 4, so I figured I'd inquire.  Thanks for posting..

UnfinishedBusiness, 2021-06-05:
Okay great, I’m working on buttoning a Rpi4 image up that all setup to boot straight into ncPilot. Once I have that done, I’ll send you a link.

sirias52, 2021-06-06:
very good project to run in RPI. Is it possible to run it in RPI3?

UnfinishedBusiness, 2021-06-10:
It may run on the RPI3, but I have not tried it. The new version of ncPilot (which does not need my previous Xkernel VM) but does still requires GL and GLFW. I've been running it on an RPI4 for about 5 months now in my shop pretty much problem-free, I've cut thousands of parts and the build-in CAM workbench (Works just like sheet cam but only supports DXF files right now) is fast on the RPI4.

closed 2021-07-14 by UnfinishedBusiness

sirias52, 2023-08-01:
HI. "Once I have that done, I’ll send you a link." can you send me de link please?